jclark/rpi-cm4-ptp-guide #1: ` nmea: unable to find utc time in leap second table` error

jclark, 2022-11-10:
Running `ts2phc -s nmea` sometimes gets the error

```
nmea: unable to find utc time in leap second table
```

jclark, 2022-11-10:
Looks like this is caused by this bug in ts2phc, which was fixed after the release of linuxptp 3.1.1

https://github.com/richardcochran/linuxptp/commit/63fc1ef4fd5e5fc45dd4de3bf27920bb109a4357?diff=unified

jclark, 2022-11-10:
Line in logs before `nmea: unable to find utc time in leap second table` is something like `nmea delay: 1668086054147466182 ns`.  This number of nanoseconds is about the difference between now and 1970. This means that here

https://github.com/richardcochran/linuxptp/blob/6feb76186ac39eab2fc011053f1cf46bfc94c8e4/ts2phc_nmea_master.c#L216

`m->rmc_time` was (incorrectly) zero.  This fields is set in `monitor_nmea_status` from the `ts` field of `struct nmea_rmc`. The failing call to `mktime` will cause this not to be initialized.

jclark, 2022-11-30:
Better link is

https://sourceforge.net/p/linuxptp/code/ci/63fc1ef4fd5e5fc45dd4de3bf27920bb109a4357/

geerlingguy, 2025-04-18:
I'm seeing this over on https://github.com/geerlingguy/time-pi/issues/15#issuecomment-2815881245 with ptp4l 4.2, I'm guessing i'm missing something else though. If you see this message, would it still wind up syncing fine otherwise, or would it mess up the time sync?

jclark, 2025-04-19:
It can also happen if the compoled-in leap second table is out of date. They have an expiry date 6 months in the future. There’s a optoon to specify a more recent one, which can de downloaded.

geerlingguy, 2025-04-19:
@jclark - Ah, thanks! I will test that out; Nvidia's PTP documentation said something similar...

https://docs.nvidia.com/networking/display/nvidia5ttechnologyusermanualv10/phc+synchronization+with+onboard+gnss#src-89130414_PHCSynchronizationwithOnboardGNSS-LeapSecondsFile