jclark/rpi-cm4-ptp-guide #10: PTP client behavior

kondou, 2023-03-20:
Today, I received additional CM4 and standard IO board.  This is for PTP client to confirm if grandmaster does its role correctly or not from the standpoint of PTP client, hence there's no GNSS related stuff attached on IO board.

I followed https://github.com/jclark/rpi-cm4-ptp-guide/blob/main/time-config.md#ptp-client-with-ntp-on-cm4, and I see this log message in syslog.

`interface 'eth0' does not support requested timestamping`

Then, I created phc@.service file as instructed in the page along with two commands.  But still I see the above log message in syslog.
Am I still missing something?  And I don't see any meaningful status for PTP0 in the result of `chronyc sources` like this.
192168.1.7 is grandmaster which has been up and running.

`root@ocs1:/home/kondou# chronyc sources`
`MS Name/IP address         Stratum Poll Reach LastRx Last sample               `
`===============================================================================`
`#? PTP0                          0   2     0     -     +0ns[   +0ns] +/-    0ns`
`^- 2a01:b740:a20:3000::1f2       1   8   377   184  +4803us[+4803us] +/-  125ms`
`^- 2003:a:47f:abe4:48ba:cd4>     2   8   377    57   -757us[ -756us] +/-  138ms`
`^- ntp3.ds.network               3   8   377   189  +2549us[+2549us] +/-   43ms`
`^- ntp.no-sense.net              2   8   373   186  -2821us[-2821us] +/-  152ms`
`^* 192.168.1.7                   1   4   377     2   -887ns[ -953ns] +/-  131us`

I don't see any PTP related messages in syslog after ptp4l started while I can see many of them on a host which is doing grandmaster role.

And `phc_ctl eth0 cmp` does not show meaningful result...

`root@ocs1:/home/kondou# phc_ctl eth0 cmp`
`phc_ctl[2884.113]: offset from CLOCK_REALTIME is 1679298645732591320ns`

I don't replace any ptp4l stuff which can be obtained from sourceforge yet.

jclark, 2023-03-20:
I think I see the problem. In `phc@.service`, change the line that says

```
Before=ptp4l@%i.service
```

to

```
Before=timemaster.service
```

kondou, 2023-03-20:
Then line 5 will be same as line 6 in the file.  Is this correct?
I just did it, but I still get the same result for `chronyc sources` and `phc_ctl eth0 cmp`...

jclark, 2023-03-20:
Sorry, I was looking at an old version of the file that didn't have the timemaster.service line.

What does

```
sudo systemctl status phc@eth0.service
```

say?

kondou, 2023-03-20:
This is before you suggested to change the line.

`root@ocs1:/home/kondou# systemctl status phc@eth0.service`
`〓〓〓 phc@eth0.service - Ensure PTP hardware clock (PHC) of eth0 is ready`
`     Loaded: loaded (/etc/systemd/system/phc@.service; enabled; vendor preset: enabled)`
`     Active: active (exited) since Mon 2023-03-20 20:14:44 JST; 20min ago`
`       Docs: man:phc_ctl`
`    Process: 447 ExecStart=/usr/sbin/phc_ctl -Q eth0 (code=exited, status=0/SUCCESS)`
`   Main PID: 447 (code=exited, status=0/SUCCESS)`
`        CPU: 4ms`
`Mar 20 20:14:44 ocs1 systemd[1]: Starting Ensure PTP hardware clock (PHC) of eth0 is ready...`
`Mar 20 20:14:44 ocs1 phc_ctl[447]: [6.559] `
`                                   capabilities:`
`                                     100000000 maximum frequency adjustment (ppb)`
`                                     0 programable alarms`
`                                     1 external time stamp channels`
`                                     1 programmable periodic signals`
`                                     1 configurable input/output pins`
`                                     doesn't have pulse per second support`
`                                     doesn't have cross timestamping support`
`Mar 20 20:14:44 ocs1 systemd[1]: Finished Ensure PTP hardware clock (PHC) of eth0 is ready.`

This is after you suggested to change the line.

`root@ocs1:/home/kondou# systemctl status phc@eth0.service`
`〓〓〓 phc@eth0.service - Ensure PTP hardware clock (PHC) of eth0 is ready`
`     Loaded: loaded (/etc/systemd/system/phc@.service; enabled; vendor preset: enabled)`
`     Active: active (exited) since Mon 2023-03-20 20:38:22 JST; 1min 21s ago`
`       Docs: man:phc_ctl`
`    Process: 473 ExecStart=/usr/sbin/phc_ctl -Q eth0 (code=exited, status=0/SUCCESS)`
`   Main PID: 473 (code=exited, status=0/SUCCESS)`
`        CPU: 9ms`
``
`Mar 20 20:38:22 ocs1 systemd[1]: Starting Ensure PTP hardware clock (PHC) of eth0 is ready...`
`Mar 20 20:38:22 ocs1 systemd[1]: Finished Ensure PTP hardware clock (PHC) of eth0 is ready.``

jclark, 2023-03-20:
When you see this line

```
interface 'eth0' does not support requested timestamping
```

in the syslog, what is the complete line?  There should also be a line nearby that has phc_ctl in it. Can you show me the section of the log that includes both the `does not support requested timestamping` and the phc_ctl line?

The underlying problem is point 5 here:

https://www.mail-archive.com/linuxptp-users@lists.sourceforge.net/msg03069.html

The phc service is supposed to work around this: phc_ctl fails if it cannot find the PHC for an interface.  Do you have the timemaster service enabled?

kondou, 2023-03-20:
Here are the syslog messages after rebooted.  Only ptp4l and phc_ctl related ones.

`Mar 20 20:38:21 ocs1 phc_ctl: [5.831] interface eth0 does not have a PHC`
`Mar 20 20:38:21 ocs1 phc_ctl: [6.366] interface eth0 does not have a PHC`
`Mar 20 20:38:22 ocs1 phc_ctl: [6.570] interface eth0 does not have a PHC`
`Mar 20 20:38:22 ocs1 timemaster: [6.579] process 442 started: /usr/sbin/ptp4l -l 5 -f /var/run/timemaster/ptp4l.0.conf -S -i eth0 `
`Mar 20 20:38:22 ocs1 ptp4l: [6.597] [0:eth0] port 1: INITIALIZING to LISTENING on INIT_COMPLETE`
`Mar 20 20:38:22 ocs1 ptp4l: [6.608] [0:eth0] port 0: INITIALIZING to LISTENING on INIT_COMPLETE`
`Mar 20 20:38:22 ocs1 ptp4l: [6.608] [0:eth0] port 1: link down`
`Mar 20 20:38:22 ocs1 ptp4l: [6.608] [0:eth0] port 1: LISTENING to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)`
`Mar 20 20:38:22 ocs1 ptp4l: [6.609] [0:eth0] selected local clock e45f01.fffe.bb1e0c as best master`
`Mar 20 20:38:22 ocs1 ptp4l: [6.609] [0:eth0] port 1: assuming the grand master role`
`Mar 20 20:38:22 ocs1 ptp4l: [6.609] [0:eth0] port 1: master state recommended in slave only mode`
`Mar 20 20:38:22 ocs1 ptp4l: [6.609] [0:eth0] port 1: defaultDS.priority1 probably misconfigured`
`Mar 20 20:38:22 ocs1 ptp4l: [6.890] [0:eth0] selected local clock e45f01.fffe.bb1e0c as best master`
`Mar 20 20:38:22 ocs1 ptp4l: [6.890] [0:eth0] port 1: assuming the grand master role`
`Mar 20 20:38:22 ocs1 ptp4l: [6.890] [0:eth0] port 1: master state recommended in slave only mode`
`Mar 20 20:38:22 ocs1 ptp4l: [6.890] [0:eth0] port 1: defaultDS.priority1 probably misconfigured`
`Mar 20 20:38:22 ocs1 phc_ctl: [6.894] #012capabilities:#012  100000000 maximum frequency adjustment (ppb)#012  0 programable alarms#012  1 external time stamp channels#012  1 programmable periodic signals#0121 configurable input/output pins#012  doesn't have pulse per second support#012  doesn't have cross timestamping support`
`Mar 20 20:38:37 ocs1 ptp4l: [10.972] [0:eth0] port 1: link up`
`Mar 20 20:38:37 ocs1 ptp4l: [10.972] [0:eth0] interface 'eth0' does not support requested timestamping mode, set link status down by force.`
`Mar 20 20:38:37 ocs1 ptp4l: [10.972] [0:eth0] selected local clock e45f01.fffe.bb1e0c as best master`
`Mar 20 20:38:37 ocs1 ptp4l: [10.972] [0:eth0] port 1: assuming the grand master role`
`Mar 20 20:38:37 ocs1 ptp4l: [10.972] [0:eth0] port 1: master state recommended in slave only mode`
`Mar 20 20:38:37 ocs1 ptp4l: [10.972] [0:eth0] port 1: defaultDS.priority1 probably misconfigured`
`Mar 20 20:38:37 ocs1 ptp4l: [10.973] [0:eth0] port 1: link up`
`Mar 20 20:38:37 ocs1 ptp4l: [10.973] [0:eth0] interface 'eth0' does not support requested timestamping mode, set link status down by force.`
`Mar 20 20:38:37 ocs1 ptp4l: [10.974] [0:eth0] selected local clock e45f01.fffe.bb1e0c as best master`
`Mar 20 20:38:37 ocs1 ptp4l: [10.974] [0:eth0] port 1: assuming the grand master role`
`Mar 20 20:38:37 ocs1 ptp4l: [10.974] [0:eth0] port 1: master state recommended in slave only mode`
`Mar 20 20:38:37 ocs1 ptp4l: [10.974] [0:eth0] port 1: defaultDS.priority1 probably misconfigured`

For timemaster service, I believe it's enalbed since I can see it via systemctl status.

`             〓〓〓〓〓〓timemaster.service `
`             〓〓〓 〓〓〓〓〓〓431 /usr/sbin/timemaster -f /etc/linuxptp/timemaster.conf`
`             〓〓〓 〓〓〓〓〓〓441 /usr/sbin/chronyd -n -f /var/run/timemaster/chrony.conf`
`             〓〓〓 〓〓〓〓〓〓442 /usr/sbin/ptp4l -l 5 -f /var/run/timemaster/ptp4l.0.conf -S -i eth0`