jclark/rpi-cm4-ptp-guide #11: PTP settings

JN19aban, 2023-03-29:
Hi,

I am trying to put some PTP port settings on ptp4l-gm config file but seems not to be working.

Example I want to decrease the logannounceinterval but seems not do anything.

How can I add these settings?

Regards.

jclark, 2023-03-29:
Thanks for this question. The current config files don't make this as easy as they could.

I suggest you create a ptp4l configuration file `/etc/linuxptp/ptp4l-gm.conf` with the settings and then in `/etc/systemd/system/ptp4l-gm.service` change the line

```
ExecStart=/usr/sbin/ptp4l -i $EXTTS_IFACE -l $LOG_LEVEL --tx_timestamp_timeout $TX_TIMESTAMP_TIMEOUT
```

to

```
ExecStart=/usr/sbin/ptp4l -f /etc/linuxptp/ptp4l-gm.conf.-i $EXTTS_IFACE -l $LOG_LEVEL --tx_timestamp_timeout $TX_TIMESTAMP_TIMEOUT
```

Then do

```
sudo systemctl daemon-reload
```

May I ask what it is you are using that needs these settings?

JN19aban, 2023-03-29:
Hi thank you for your response.

It will be used for SMPTE environment and they are really helpful.

I have done that one you have told me and is working perfect. Although maybe is better to avoid completely the rest settings on the ptp4l-gm config and have it all on the  ptp4l-gm.conf? 

An other thing is maybe the CPU governor put it on Performance mode for having better response and lower latency?

BTW I appreciate your work.

Regards.