jclark/rpi-cm4-ptp-guide #12: GPS strange behavior

JN19aban, 2023-03-31:
Hi,

I am noticing a strange behavior on the GPS module I have for the test. 

When I power off the CM4 and leave it for about 5 minutes off and plug in the power the GPS is stalled no connection nothing. Now if remove the power and power on again it start working.

I have partially resolve this by remove the pin RXD on GPS. After that not an issue no many power cycles I do.
But it has be connected to be normal. 

Is there any config that should do either to GPS or CM4?
----------
And an other issue is that with the sernet can not connect via the ublox app on Windows. I have enter on the app tcp://192.168.2.142:2002 which .142 is the CM4 ip address but can not connect. Even with ts2phc stopped no response still.

Regards.

jclark, 2023-03-31:
What GPS module are you using?

JN19aban, 2023-03-31:
Ublox NEO6-m it is original, I have buy it from authorized dealer.

jclark, 2023-03-31:
NEO6m should work straightforwardly.

Do you have a serial to USB adapter?  If so, try get it working with that first.

Does

```
(stty 9600 -echo -icrnl; cat) </dev/ttyAMA0
```

give you output?

Double-check to make sure you haven't got two processes reading the serial port? Several times I have had problems and it turned out I had accidentally done this.  In particular, make sure gpsd isn't running.

jclark, 2023-03-31:
> And I am saying it is stalled because the PPS led is constantly on. I have left it above 24 hours like this way and is still halted, but if i powercycle it blinks after ~5-10 seconds.

Blinking after 5-10s is what I would expect.

If the PPS led is on all the time, I would suspect some sort of electrical problem.

JN19aban, 2023-03-31:
All are fine with cables etc. Strange but the issue is with the RXD pin. I have connected it either with 3.3v or 5v (it accepts both) and do the same thing.

JN19aban, 2023-04-02:
After all it appeared to be the GPS module. I replace it with a new one and the problem solved.

closed 2023-04-02 by JN19aban