jclark/rpi-cm4-ptp-guide #13: parameter setting for grandmaster

kondou, 2023-04-02:
I was trying to change some of parameters, like priority1 or priority2 for grandmaster, and I though it would be done at /etc/default/ptp4l-gm at first, but nothing was changed by seeing ethernet data via wireshark.
I also tried to change /usr/local/sbin/ptp4l-gm-set.sh, but this didn't also look like the right one to be modified.

Which file do I have to modify to change the default parameter for ptp4l?

jclark, 2023-04-02:
See #11

kondou, 2023-04-02:
Thank you very much.  I thought I read that one, but totally forgot.

For my case, the original line was slightly different from #11.
`ExecStart=/usr/sbin/ptp4l --masterOnly 1 -i $EXTTS_IFACE -l $LOG_LEVEL --tx_timestamp_timeou
t $TX_TIMESTAMP_TIMEOUT`
And I modified like this.
`ExecStart=/usr/sbin/ptp4l -f /etc/linuxptp/ptp4l-gm.conf --masterOnly 1 -i $EXTTS_IFACE -l $L
OG_LEVEL --tx_timestamp_timeout $TX_TIMESTAMP_TIMEOUT`
I copied `ptp4l.conf` to `ptp4l-gm.conf`, and changed related lines.

I noticed that the log file at slave would never include like this messages after I modified priority1 and 2.
`Apr  3 00:41:31 ocs1 ptp4l: [441929.796] [0:eth0] selected best master clock e45f01.fffe.bb1fc8`

Now I assume this line would be logged if there are multiple grandmaster on the same network with the same priority.
Unless otherwise, these three lines look like the last message logged by ptp4l.
`Apr  3 00:41:41 ocs1 ptp4l: [441940.283] [0:eth0] selected best master clock e45f01.fffe.c90cae`
`Apr  3 00:41:41 ocs1 ptp4l: [441940.283] [0:eth0] port 1: SLAVE to UNCALIBRATED on RS_SLAVE`
`Apr  3 00:41:43 ocs1 ptp4l: [441942.284] [0:eth0] port 1: UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED`