jclark/rpi-cm4-ptp-guide #14: Serial port

JN19aban, 2023-04-02:
After I have replaced the module I did a complete refresh on CM4 all ok so far.

But now the /dev/ttyAMA0 is not visible anymore, instead is serial0 or /dev/ttyS0. Is that an issue if I leave it like this?

My CM4 has not wifi built in, only EMMC.

jclark, 2023-04-03:
The serial port setup is rather complicated. The full details are here: https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-uarts

It will probably work ok as is.

You need this in `/boot/config.txt`

```
# Make /dev/ttyAMA0 be connected to GPIO header pins 8 and 10
# This always disables Bluetooth
dtoverlay=disable-bt
```

And also enable serial as described in https://github.com/jclark/rpi-cm4-ptp-guide/blob/main/os.md#os-configuration

JN19aban, 2023-04-03:
OK thank you by this way it the ttyAMA0 is showed up.

`Reference ID    : 53484D30 (SHM0)
Stratum         : 1
Ref time (UTC)  : Mon Apr 03 03:19:03 2023
System time     : 0.000000227 seconds fast of NTP time
Last offset     : -0.000000071 seconds
RMS offset      : 0.000001926 seconds
Frequency       : 12.926 ppm fast
Residual freq   : -0.000 ppm
Skew            : 0.037 ppm
Root delay      : 0.000000001 seconds
Root dispersion : 0.000006802 seconds
Update interval : 14.0 seconds
Leap status     : Normal
`

The RMS still not dual or single digit. It can be from the performance of NEO6?

jclark, 2023-04-03:
It takes quite a long time for the RMS to settle down. What sort of offsets is ts2phc logging?

NEO6M is quite old now, but I would expect ts2phc to be able to keep the PHC within much less than 100ns of the time pulse.

JN19aban, 2023-04-03:
`Apr 03 04:15:49 raspberrypi ts2phc[489]: [3630.100] nmea delay: 144732792 ns
Apr 03 04:15:49 raspberrypi ts2phc[489]: [3630.100] eth0 extts index 0 at 1680495385.999999992 corr 0 src 1680495386.865405308 diff -8
Apr 03 04:15:49 raspberrypi ts2phc[489]: [3630.100] eth0 master offset         -8 s2 freq   +2724
Apr 03 04:15:49 raspberrypi ts2phc[489]: [3630.236] nmea sentence: GPRMC,041549.00,A,3801.85724,N,02121.96150,E,0.321,,030423,,,A
Apr 03 04:15:49 raspberrypi ts2phc[489]: [3630.352] nmea sentence: GPGGA,041549.00,3801.85724,N,02121.96150,E,1,08,1.05,25.7,M,32.7,M,,
Apr 03 04:15:49 raspberrypi ts2phc[489]: [3630.402] nmea sentence: GPGSA,A,3,30,07,15,14,05,20,13,17,,,,,2.02,1.05,1.73
Apr 03 04:15:49 raspberrypi ts2phc[489]: [3630.469] nmea sentence: GPGSV,3,1,11,05,43,241,23,07,13,072,31,08,00,028,,13,62,324,28
Apr 03 04:15:49 raspberrypi ts2phc[489]: [3630.552] nmea sentence: GPGSV,3,2,11,14,69,073,32,15,33,311,49,17,19,132,25,19,06,157,18
Apr 03 04:15:49 raspberrypi ts2phc[489]: [3630.602] nmea sentence: GPGSV,3,3,11,20,27,211,16,24,03,270,,30,46,058,37
Apr 03 04:15:49 raspberrypi ts2phc[489]: [3630.652] nmea sentence: GPGLL,3801.85724,N,02121.96150,E,041549.00,A,A
Apr 03 04:17:13 raspberrypi ts2phc[489]: [3714.284] nmea delay: 146807327 ns
Apr 03 04:17:13 raspberrypi ts2phc[489]: [3714.285] eth0 master offset         14 s2 freq   +2734

`

A good alternative would be M9n M8t for example?

jclark, 2023-04-03:
This looks fine: you have offsets of 14ns and -8ns.

M8T gives you timing mode, which should help a little, but it's not going to be a huge difference. A good antenna (in a good location) is just as important.

JN19aban, 2023-04-03:
It is an external original ublox one with ground plane installed.

https://www.u-blox.com/en/product/ann-mb-series

JN19aban, 2023-04-03:
To something more to ask the ts2phc "automatic" take PPS from GPS module on the pin on CM4?
Not need extra config?

kondou, 2023-04-03:
Any of M8F, I have 3 sets, spews `master sync timeout` while I also see `master offset ... s2 freq ...`, which is no difference from my situation.  These look like what is expected from my one month experiences.

JN19aban, 2023-04-26:
A small update.

I have somehow managed to connect to Gps Module via ethernet but with some limitation. When disable ts2phc I can connect and control the module via u-blox app, this with ser2net software. When ts2phc is running can not be connected or have very slow response. 

Is that normal? I know the ts2phc uses the serial but can be working the same time?

kondou, 2023-04-26:
When I need to connect Ublox module to run u-center, I always stop ts2phc and start ser2net.  I've never connected with ser2net while ts2phc was running.

jclark, 2023-04-26:
ts2phc and ser2net can't work at the same time. They would each be seeing some random part of the output from the U-blox module.

JN19aban, 2023-04-26:
Thank you I appreciate both for the answers.

closed 2023-04-26 by JN19aban