jclark/rpi-cm4-ptp-guide #15: MOTU AVB switch as TC

kondou, 2023-04-03:
It looks like two types of MOTU AVB switch and both are slightly different, but I'm not sure if there is any differences in its functionality.

[image]
[image]

And I expected MOTU AVB switch behaves as TC and correction field in PTP data should have meaningful data, but seeing from the ethernet data via wireshark, correction field is always all zero.  The result of tcpdump at slave shows all zero correlation data from GMC.  All devices are connected to MOTU AVB switch.  I have both types MOTU AVB switch, and I cannot see any data other than zero in correlation field for both type.  The firmware of the switch is the latest one.

Do i misunderstand what TC does, or am I missing something?

jclark, 2023-04-04:
It would be great to be able to get this working. I don't know much about this...

I am guessing that the problem here is that it is specifically an *AVB* switch rather than a general-purpose IEEE1588 switch, and that this means it will only work when IEEE1588 is being used in an AVB-compliant way.

AVB includes the 802.1AS profile of  IEEE1588, which is also called gPTP, plus a bunch of other stuff. I see there's a file `/usr/share/doc/linuxptp/configs/gPTP.cfg`, so I think the first thing to try is using that config.

kondou, 2023-04-04:
Thank you very much for the comment.

At first, I tried to run ptp4l with -2 while the configuration file was not changed, but it looked like no PTP related ethernet data sent to the network.
Then I tried to use gPTP.cfg as the configuration file without -2 option, but ptp4l didn't send anything related to PTP.
I ran tcpdump at PTP slave which was prepared to monitor how PTP slave behaves to see if any PTP ethernet data can be received.

Seeing from the situation, MOTU AVB switch may not be suitable for the current environment.  This means there is no TC in my network environment, and I may not expect sub micro precision from GMC.  Then, I may need to connect PTP slave directly to GMC without L2 switch...

jclark, 2023-04-04:
Looks like AVB also needs some VLANs and traffic control stuff. See https://tsn.readthedocs.io/avb.html#system-setup 

But, as you say, if Dante is expecting UDP transport rather than L2, then I guess it's going to be difficult to make it work.

I just got an IES3110-8TF-R as well, but I haven't set it up yet. There's some updated FW here: https://www.fs.com/sg/products_support/search.html?id=148180  (which the IES3110-8TF doesn't). the  The CLI manual has a bunch of new PTP stuff.  I think it's likely the best option.

kondou, 2023-04-04:
Thank you very much for the information.  Now, I give up MOTU AVB for my usage.

I've just ordered IES3110-8TF-R, which expects be delivered middle of June.  Until I receive them, I try to be more familiar to PTP operation and its behavior.

p.s.
For GNSS, I found how to enable QZSS a few days ago.  I did survey several times changing where the antenna is located in the veranda, but it still didn't provide good result.  I decreased the number of GPS to be used to avoid SVs which looked like observed via reflected radio wave via some of tall buildings around my house, and did survey again for short period.  This provided much better result than what I've done many times.  After survey, it started fixed mode, and I decreased more of GPS so that QZSS would be referred much more.  This may be tricky, but I'm going to see how it will look like.

kondou, 2023-04-06:
Thank you very much for the information.

I used to use SG300, not 350, for daily work, it's too noisy (fan noise) for private use.  Even for the used one, it looks much expensive in Japan than what James suggested (feisu.com aka fs.com).

JN19aban, 2023-04-06:
The 10 port versions have no fan as the biggest models but the fs are nice too.