jclark/rpi-cm4-ptp-guide #16: When reboot the CM4 the system halt

JN19aban, 2023-04-12:
Hi,

I am try to solve an issue but I haven't success yet. When everything operate smoothly and do a normal reboot sometimes the CM4 hangs can not connect via ssh or send PTP packets. To resolve this I need to do a powercycle. It does not hang all the time but some off them.

Any idea what might be cause this?

Regards.

jclark, 2023-04-12:
Does it hang while shutting down or while starting up again?

I think I've seen it hang like this during the shutdown process, but it hasn't happened to me recently and I don't think I ever found the cause.  My guess is that one of the daemon's is hanging when systemd tries to kill it. So the thing todo would be to look in the logs to find out which daemons successfully shutdown and which don't.

JN19aban, 2023-04-12:
This happening only on startup and not on shutdown. I am suspecting the the ts2phc it might start earlier than or the same time that the driver enables the link on phc. I have checked the log files but nothing,

Let me ask can on ts2phc ptp4l to add some delay on startup just to check this thing?

Regards.

jclark, 2023-04-12:
The way I've set it up is that ts2phc (and the other daemons) are started by a dhcpcd hook when a carrier is detected.  This means they won't get started until the the ethernet port is up and running.

See /etc/dhcpcd.exit-hook (which you can modify as you like)

Make sure ts2phc is disabled (so that it's not started by systemd).

kondou, 2023-04-12:
Just fyi.

> I think I've seen it hang like this during the shutdown process, but it hasn't happened to me recently and I don't think I ever found the cause. My guess is that one of the daemon's is hanging when [...]

I've experienced this situation also, and I connected display to see what is happening when CM4 looked up but cannot login via ssh.
This is what I saw.  I just need to wait 90 sec at most if this kind of situation happens again.
I would not do the power cycle by plugging out the power cable, because IO board looked like getting certain damage in my case, and CM4 suddenly rebooted several times.  I had to replace IO board with the new one, which fixed the unexpected sudden reboot.
[image]

JN19aban, 2023-04-26:
Solved this issue with the latest stable kernel `6.1.21`

closed 2023-04-26 by jclark