jclark/rpi-cm4-ptp-guide #17: configuring IES3110-8TF-R as TC

kondou, 2023-04-19:
Hi,

Yesterday, I managed to obtain IES3110-8TF-R at last, and am trying to configure it as TC.
Anyone successfully has configured it?  I've been fumbling even with the guide(ies3110-8tf-and-ies3110-8tf-p-switches-configuration-guide.pdf).  There's no exact explanation in the guide about the profile...  I tried one by one, but it does not work as TC.  I might not have done something correctly...
[image]

kondou, 2023-04-19:
Thank you very much for the information.
I tried the same thing as you did, and now Clock Adjustment method is changed to Internal Timer from Software Timer.
But the PTP Time is randomly changing.  And it looks like GMC cannot be found on the switch.  I tried to set false Not Slave column for the port where GMC (CM4) is connected, but no difference...  Also tried to decrease the Local Prio but still no change...  Correction field in the packet is always zero by seeing at PTP slave via tcpdump...  This might be due to the different behavior from 8TF.
[image]

jclark, 2023-04-19:
Is the Protocol right? `Ethernet` is for the ptp4l L2 transport. Mine is set to `IPv4Multi`.

jclark, 2023-04-19:
The PTP support is clearly still a bit rough. The UI has a whole bunch of stuff that only makes sense when it is being used as a boundary clock, but it doesn't support configuration as a boundary clock.  I am hoping that the  IES3110-8TF-R with the latest firmware can be made to work as a boundary clock (at least via the CLI).

kondou, 2023-04-19:
Then, you're going to use as BC instead of TC, aren't you?  That might be one of options for me also.

FWIW, my 8TF-F is V1.1 which came with when shipped, and what FS sales engineer guy, today, told me that 1.1 would be the latest one.