jclark/rpi-cm4-ptp-guide #20: nmea: utc time is past leap second table expiry date

kondou, 2023-06-28:
Since from 00UTC today, ts2phc complaints like this.  All three grandfmaster fell into this situation at the same time.  This happens only to me?
[image]

jmfriedt, 2023-07-08:
Also happened to me as Buildroot generated linuxptp on a Compute Module4 did not have /usr/share/zoneinfo, which I copied from Debian/GNU Linux desktop. Check the content of the file /usr/share/zoneinfo/leap-seconds.list with its expiry date as
```
#       Updated through IERS Bulletin C64
#       File expires on:  28 June 2023
#
#@      3896899200
```
Just increase the expiry date or fetch https://www.ietf.org/timezones/data/leap-seconds.list for the updated version.

JN19aban, 2023-07-08:
This is strange because on my CM4 have the correct leap-seconds.list with expiry date `28 December 2023`

kondou, 2023-07-16:
Sorry for a bit late reply, but that leap-seconds-list file works perfectly for me, and no restart was required even after that file was replaced.
Thanks.

closed 2023-07-16 by kondou

jclark, 2023-07-17:
The whole leap second business with ts2phc is so sub-optimal:  ublox receivers can provide the time in TAI, as well as the leap second offset.

kondou, 2023-08-05:
I need to understand how to retrieve the time in TAI from ublox...

jclark, 2023-08-05:
@kondou The UBX-NAV-TIMEGPS message provides this. I have written a program that takes advantage of this. The program is in a private repository currently, because there are a few things I want to do before I release it publicly. But if you would like to give it a try, I can give you access to the repo.

kondou, 2023-08-05:
Thank you very much for the proposal of the trial.  At the moment, the issue should not happen for the time being, and I have different projects to be done which is completely different from PTP.  So, I'm okay after the public release.  But, I may ask you when the priority of those projects is getting down.