jclark/rpi-cm4-ptp-guide #22: Add info about Mikrotik switches

jclark, 2023-08-05:
Some Mikrotik switches have PTP support.

There's a list at the bottom of this page https://help.mikrotik.com/docs/display/ROS/Precision+Time+Protocol

jclark, 2023-08-05:
The cheapest is the CRS326-24G-2S, which has two variants:

- CRS326-24G-2S+RM, which is rack-mounted
- CRS326-24G-2S+IN, which is desktop

I just got the RM version

https://mikrotik.com/product/CRS326-24G-2SplusRM

List price is only $209.  It seems to be widely available: I was able to get from a local retailer in Thailand.

The ports are

- 1Gb RJ45 x 24
- 10Gb SFP+ x 2

As regards PTP support

* only the RJ45 ports have PTP support
* works as boundary clock
* does not support working as a transparent clock
* all ports have to have the same PTP profile
* has support for 802.1as

jclark, 2023-08-05:
@kondou you might be interested in this

kondou, 2023-08-05:
Thanks.  But it has no TC capability.  And it's not so cheaper than what fs.com provides.

commit e7a345ae (2023-08-21): Add information about Mikrotik switches

Fixes #22
closed 2023-08-21 by jclark