jclark/rpi-cm4-ptp-guide #23: LinuxPTP 4.0

Doridian, 2023-08-09:
Fairly recently, LinuxPTP 4.0 launched, which broke due to the CM4's (apparently) broken PTP implementation.
It took me hours to realize this, and then even more to fix it.

In case anyone wants to run 4.0 on their CM4, here's the patch your need: https://github.com/FoxDenHome/ntp/blob/main/input/src/linuxptp-minor-version-2-0.patch

(Essentially, a common theme with broken PTP hardware seems to be it enforcing a specific version on the packets while it SHOULD be tolerant to minor version changes, but isn't, so this changes the minor back to 0 allowing everything to function)

jclark, 2023-08-10:
There was some discussion of this on the linuxptp mailing list:

https://www.mail-archive.com/linuxptp-users@lists.sourceforge.net/msg03316.html

There's also a patch that adds an option to make the version configurable:

https://www.mail-archive.com/linuxptp-devel@lists.sourceforge.net/msg06374.html

But the linuxptp maintainer has declined to merge  it.  So probably  we should package version 4 with this patch at least for CM4.

Doridian, 2023-08-10:
> There was some discussion of this on the linuxptp mailing list:
>
> https://www.mail-archive.com/linuxptp-users@lists.sourceforge.net/msg03316.html
>
> There's also a patch that adds an option to make the [...]

Yeah, those threads are how I found out at all that this is a thing that happens. Then I found the rejection of the patch, so I made my own patchfile and put it here so people don't have to follow my descent all the way down the rabbit hole (even though...if you've arrived at deploying PTP in your home network, you are already quite deep in the timing rabbit hole, soooo)

jclark, 2023-12-28:
I've developed a solution to this using BPF, which does not require patching LinuxPTP 4.0.

https://github.com/jclark/bpf-ptpver20

jclark, 2024-04-14:
This is now fixed upstream in LinuxPTP 4.2 (https://github.com/richardcochran/linuxptp/commit/02a13b039d44479ba429c3cc1b75acb8e6da924a)

You can say `ptp_minor_version 0` to make this work again. We should document this.