jclark/rpi-cm4-ptp-guide #25: LEA-M8F PPS issue

JN19aban, 2023-09-06:
Hi,

I have upgraded from NEO-6m module to LEA-M8F and I am facing an issue. When I start the CM4 the PHY is not working although the NIC leds are on.

After troubleshooting a bit I am seeing that if I disconnect the PPS cable or disable the PPS from the U-center the CM4 operates normally.

If I connect most of the times it loses the PHC.

The settings on TP5 are below:

[image]

Any help is appreciated.

Regards.

jclark, 2023-09-06:
Are you using the TimeBeat module?

When you say `disconnect the PPS cable` what do you mean?  You don't need a PPS cable.

JN19aban, 2023-09-06:
No it is the module from GNSS store, with the usb one.

This happened when I reset the settings on default on U-center.

jclark, 2023-09-06:
How exactly do you have it wired up? I'm assuming pins 1 and 3 on M8F board are connected to GND and SYNC_OUT on the CM4 board. Do you have USB connected to the PC running U-center or are you connecting RX/TX/power to the CM4?

Is the PPS LED on the M8F board blinking when PPS cable is connected? Is it blinking when the PPS cable is not connected?

Have you tried attaching an oscilloscope to the PPS output from M8F?

I have an M8F from TimeBeat. I haven't done anything with it yet, except to check that I am getting a PPS signal and NMEA data (which I am). With the TimeBeat, the daughter board handles the connection to SYNC_OUT, but that shouldn't make any difference at the software level.

Looking at the receiver description https://content.u-blox.com/sites/default/files/products/documents/u-blox8-M8_ReceiverDescrProtSpec_UBX-13003221.pdf , the FTS range of modules (i.e. the M8F) has its own chapter (chapter 26). But that seems it uses TP5 as usual.

jclark, 2023-09-06:
If you're connecting the M8F to U-center using a USB connection to a separate PC, then I would suggest instead connecting the TX/RX/power of the M8F board to the CM4 and then using ser2net to use U-center.  It sounds a bit like a grounding problem to me. If you are connecting with a mains-powered PC, you could also try connecting with a laptop instead.

JN19aban, 2023-09-06:
Correct, 1 and 3 GND and SYNC_OUT.

The connection on CM4 is not connected via usb at all.

The PPS cable (Timepulse) is blinking with or without connect to CM4.

Now a quick test I have done.

On U-center TP5 section when I put on second tab from the beginning of the list the Length at 0 us is working correct. And this with U-center v23.08 which on default has it on 0us.

On U-center v22.07 the defaults are 100000us which make crazy the CM4.

Now what is the correct value? On Timebeat module what is the default value on this?

Regards.

jclark, 2023-09-06:
There's something really weird going on with U-center:

[image]

JN19aban, 2023-09-06:
The same also does here on my side.

To fix it disable active on both Timepulse 0 and 1 save and then enable both and save. It should be fine. 

It is probably bugged this section on U-center.

jclark, 2023-09-06:
Maybe the explanation is this (from 26.6 of the manual):

> In current FTS products only TIMEPULSE 2 can be used for pulse generation.
> Additionally, just 0.5 Hz, 1 Hz and 2 Hz time pulse output is supported by current FTS
> products. Other output frequencies may be configured with UBX-CFG-TP5 but are not
> guaranteed to work properly.

JN19aban, 2023-09-06:
Ok, I think we need 1hz for timepulse (1000000us) period but for length 100ms or 100us?

jclark, 2023-09-06:
If I select TIMEPULSE2 in the dropdown and then Poll, I see this:

[image]

So length locked is 100000ms (0.1s)

JN19aban, 2023-09-06:
The same I have with U-Center v23.08 so this might be it. But on my side the poll is not working at all. 

I will write on u-blox forum about this issue. 

With this settings the CM4 PHY is not get "nuts" and work ok. 

At least is working correct now.

Also I am seeing with this module offsets on ts2phc from -4 to 7 (with SBAS turned off) which is a very good upgrade from 6m. If you want you can  add this on working GPS modules. 

Best regards and really appreciate your help.

closed 2024-04-10 by JN19aban