jclark/rpi-cm4-ptp-guide #27: Can the GNSS module for PTP be used for data acquisition too?

mcamurri, 2023-09-15:
I'm looking into implementing a grandmaster clock using the CM4 + IO Board + [Ardusimple RTK2B](https://www.ardusimple.com/product/simplertk2b/) (i.e. ublox F9P).

Because the CM4 has several UARTs, and the F9P has two, I was wondering if I can use the device to both feed the ptp server and also pull data from it, or if I need two devices.

TLDR

In particular, I'd need to run [this ROS wrapper](https://github.com/HKUST-Aerial-Robotics/ublox_driver/blob/main/src/ublox_driver.cpp) which accesses the serial port, takes raw measurements and sends them over the ROS network via TCP/IP. 

The way the ROS wrapper is used normally, is to just plug the device to the computer via USB and the wrapper would take the data off the serial `ttyACM0`.

I'm trying to understand if there's a hardware way to use only one device for both applications (ptp grandmaster and ROS driver) without rewriting code, e.g. by modifying the driver to take data off a `gpspipe` as suggested [here](https://lists.nongnu.org/archive/html/gpsd-users/2022-09/msg00047.html) or using some interprocess communication like shared memory.

For example, I was wondering if the same serial could be accessed by both the ROS wrapper and `gpsd` or if there's a way to plug both UARTs from the FP9 to the raspberry Pi and use one for PTP and the other for the ROS wrapper.

Any ideas?

jclark, 2023-09-15:
Looking at 

https://www.ardusimple.com/simplertk2b-hookup-guide/

it seems like that board gives you access to both the F9P UARTs and the USB port.  You can then configure the F9P to say which protocols (NMEA/UBX/RTCM) it should output on which ports; the manual says which are allowed.

If you want to use two UARTs, you would need to use a device overlay to enable an additional UART on the CM4.

https://www.raspberrypi.com/documentation/computers/configuration.html#configuring-uarts

e.g. 
```
dtoverlay=uart4
```

would put TX/RX on GPIOs 8/9.

I am using the F9P from ArduSimple in an M.2 form factor on a PC with an i225, and it works nicely for timing purposes.

mcamurri, 2023-09-15:
Thanks for the prompt answer! 
I know the ardusimple M.2 card exists and it looked like a very attractive solution, but I was skeptical to use it because it wasn't clear to me what signals are brought from the card to the computer. I couldn't find a good schematic that shows all the signals ported from the FP9 to the computer and on what PCI buses. 

For example, to work on a regular computer, the PPS signal should be wired to a specific pin of a serial port, but it it wasn't clear to me how this would be implemented on the PCI card. 

I kinda of assumed the M.2 card was just using one USB lane, effectively achieving the same as plugging the device to a USB port, but wasting a lot of PCIe pins.

Note that there exist a compact carrier board with a M.2 slot available [here](https://store.oratek.com/products/tofu?variant=42672305275078) although I haven't checked if the keys correspond.

mcamurri, 2023-09-15:
> The M.2 card only exposes the USB port, so it wouldn't work well for your application. I don't think it would be a good option for the CM4, but it's nice with a PC.
> 

If it's just USB, how can you make it work with PTP then? You don't have a pps coming into the computer. You just use it to adjust the system clock and let your intel card propagate the system's clock time to other computers by using phc2sys from the system clock to the ethernet card?

> > I thought the USB port and the UART0 were mutually exclusive
> 
> I don't remember seeing anything about that limitation, but it might be the case: I haven't tried. It should be documented in the F9P manual. I have found ArduSimple quite helpful: they are probably the best people to ask. Please report back what you find.

I'll do, thanks!

jclark, 2023-09-15:
Sorry, I should have been clearer: the only interface it exposes *for messages* is USB. The card has two U.FL connectors, one is for the antenna and the other is a PPS out, which works as expected.

mcamurri, 2023-09-15:
So you wired the PPS from the U.FL  to the carrier detect pin of a serial port on the computer, as suggested [here](https://docs.kernel.org/driver-api/pps.html))?

jclark, 2023-09-15:
I am using the Timebeat mod to the i225-T1: https://store.timebeat.app/products/intel-i225-t1-with-sdp-headers-and-a-u-fl-breakout-board

Then I used a U.FL-U.FL cable to connect the PPS pin from the M.2 Z9P card to the SDP pin on the breakout board.

[image]

mcamurri, 2023-09-18:
I understand now, thanks.

I was convinced you were using a computer with a native i225 NIC, such as a NUC computer.

I know this is getting a bit OT, but with a computer with a native Intel NIC, what would the approach be?

If I understand correctly, your approach in a nutshell is as follows:
- Drive the PHC pulse with the PPS from the receiver
- Use `ts2phc` to take the NMEA timestamps from GPS to the PHC
- use `phc2shm` to provide access to this data to chrony, who will update the system clock

The above seems not possible if the PHC sync pin is not exposed, so I was thinking of the following:

- Take the PPS from the GPS and wire it to a serial port via [linuxPPS](https://docs.kernel.org/driver-api/pps.html)
- Take the NMEA from GPS to another serial port or USB port
- Use the above two in combination with `gpsd` to create a Stratum 1 timesource for chrony ([link](https://wiki.alpinelinux.org/wiki/Chrony_and_GPSD))
- Use `phc2sys` in the opposite direction to sync the PHC clock to the system clock, which is kept in sync with GPS by chrony

Does it make sense or you think there's a simpler solution? 

**EDIT**
I just saw that `phc2sys` accepts an external PPS source as well ([link](https://linux.die.net/man/8/phc2sys))

jclark, 2023-09-18:
> Use ts2phc to take the NMEA timestamps from GPS to the PHC

What the OS does is tell you the PHC time at which a pulse occurred. The key thing that ts2phc does is to use this to synchronize the PHC time to the pulse. The NMEA data is not the heart of it: you can use NTP just as well.

I think yojr approach makes sense provided what you are interested in is accurate synchronization of the computers on the LAN relative to each other. In this case, get a system that supports PTM (i.e. get an i226-V with 12th gen chipset), so the system time and PHC can be synchronized accurately.

But if you are interested in absolute synchronization to UTC, I think you are going to lose so much accuracy going via the serial port that there is no point in using PTP: keep it simple and use NTP.

jclark, 2023-09-22:
I have now release a companion guide for doing PTP on PC hardware, which deals with the i225

https://github.com/jclark/pc-ptp-ntp-guide