jclark/rpi-cm4-ptp-guide #28: Kernel RT or LL question.

JN19aban, 2023-10-25:
I have a simple question and I hope if someone can help with that.

I am wondering if an RT or LL Kernel would help the performance of PTP or would be the same with the generic kernel.

So far from what I am seeing with LL kernel and CPU Governor on Performance has almost the same response. The only difference is on client device that the deltas seems to be a bit better.

The LL kernel build by kdoren and can be found on below link:

https://github.com/kdoren/linux/releases/tag/rpi_6.1.54-rt15

If someone has try CM4 with RT or LL or if need that kernel please share the experience.

Regards.

closed 2024-04-10 by JN19aban

jclark, 2024-04-14:
I don't think it would make any difference as regards the PTP hardware clock, because that is all based on hardware timestamping.  It might possibly make a difference for how accurately the system clock can be synchronized to the PTP hardware clock.

JN19aban, 2024-04-14:
With LL or RT kernels on RPI 4B which has not hardware timestamp, yes with RT kernels everything was way better specially the delay response. But so far on CM4 with hardware timestamp I haven't see any improvement, I didn't test / try it that much but nothing highly noticeable seen.

jclark, 2024-04-14:
I've never tried PTP with software timestamping.  Do you get significantly better synchronization than with NTP?

JN19aban, 2024-04-14:
Yes, but it need highly modification of kernel and tune the system for performance.

All these with 5.10 and 5.15 kernels. I do not know with the latest kernels if that is possible anymore.