jclark/rpi-cm4-ptp-guide #31: Raspbian 12 (Bookworm) start GM services on boot not working.

JN19aban, 2024-04-10:
HI,

I have installed the Raspbian 12 (Bookworm) on a new CM4 setup followed the instructions and on reboot (or power on) the service are not auto starting.

I have tried on Raspbian 11 (again) and work as supposed to be.

Probably this is due Network Manager on Raspbian 12?

A bit help would be appreciated.

Best regards.

JN19aban, 2024-04-11:
I have found a solution for testing it by disabling the `Network Manager` and install the `dhcpcd`.

1. sudo systemctl disable NetworkManager
2. sudo systemctl stop NetworkManager
3. sudo apt-get update
4. sudo apt-get install -y dhcpcd5
5. sudo reboot

By doing this the `dhcpcd hook` is working and on boot the services are starting and everything working normal. 

This is test so far and I do not know if it breaks anything else by disabling the `Network Manager`.

jclark, 2024-04-14:
Thanks. I think your proposed fix is the simplest way to get things working again on bookworm. Can you check

```
sudo systemctl status NetworkManager-wait-online.service
```

? That should be disabled if NetworkManager is disabled.

Some of the desktop stuff will work better with NetworkManager, but I don't think that matters much for this use case.

JN19aban, 2024-04-14:
Yes I can confirm that is disabled. 

The output:

`NetworkManager-wait-online.service - Network Manager Wait Online
     Loaded: loaded (/lib/systemd/system/NetworkManager-wait-online.service; disabled; preset: enabled)
     Active: inactive (dead)
`

So far it seems to be working normal, in any case I will update.

JN19aban, 2024-12-18:
I think I have made a script for Network Manager hook which start the ptp4l-gm and phc2shm on PHY **up** state and on **down** state stop the ts2phc. The script is based on the **dhcpcd.exit-hook**. You should put the script for test on /etc/NetworkManager/dispatcher.d/nm-hook and make it executable by using the command **`chmod +x nm-hook`**.

The script:

```
#!/bin/bash

# Log file for debugging
LOGFILE="/var/log/nm-hook.log"

# Extract arguments
INTERFACE="$1"
STATUS="$2"

# Ensure only the target interface is processed
if [ "$INTERFACE" == "eth0" ]; then
    case "$STATUS" in
        "up")
            echo "$(date): Interface $INTERFACE is UP. Starting services..." >> $LOGFILE
            systemctl start ptp4l-gm phc2shm
            ;;
        "down")
            echo "$(date): Interface $INTERFACE is DOWN. Stopping ts2phc..." >> $LOGFILE
            systemctl stop ts2phc
            ;;
        *)
            echo "$(date): Unknown status $STATUS for interface $INTERFACE." >> $LOGFILE
            ;;
    esac
else
    echo "$(date): Ignoring interface $INTERFACE with status $STATUS." >> $LOGFILE
fi
```

@jclark try this and see if it working ok, on my side everything is fine.

Best regards,
Jim

jclark, 2024-12-20:
I am away at the moment, so I can't test, but this looks like a promising approach. I didn't know about dispatcher scripts in NetworkManager.

JN19aban, 2024-12-20:
OK,

on my side it seems to be working smoothly. The most easier and promising is the **dispatcher.d** that do all the job, will see.

Although I am seeing some performance improvement at least on whole system. 
**An other thing that I am seeing is that the **ser2net** it auto creates a service and is already enabled when installing it by default.**

I keep watching for issues. 

Regards.

hagak, 2025-01-04:
JN19aban  - thanks your script worked great for me.

JN19aban, 2025-02-12:
@jclark When you have time fix a bit the documentation for Bookworm add the new script above, etc. and inform for CM5.....

commit d12260ed (2025-02-13): Update ptp4l-gm docs for NetworkManager

Fixes #31
closed 2025-02-13 by jclark