jclark/rpi-cm4-ptp-guide #32: Can the rasp pi 5 provide the GPS locked time server with PTP ?

g4tsn, 2024-04-27:
Hi  Question 
Does anyone know if the Pi 5 can be used to provide the GPS locked time server with PPS to GPIO and transmitting the PTP via network to several clients using this project with cm4 to supply PPS signal out to various devices for time sync ?   

I believe from reading the Pi 5 can do the hardware time stamp as server, but not as client create the PPS to GPIO ....so clients would need to use cm4 as they have the sync pin out ?   doesn't look like its possible to use a pi 5 as a client and output the network sync PPS via a GPIO ?  

Can someone confirm this please I just want to decide to order some cm4 boards or pi 5's 

Thanks 

Jon

jclark, 2024-04-28:
I'm sorry but I don't understand what you are trying to do.

The CM4 will work much better than the Pi 5 for syncing PPS input or output with PTP.

g4tsn, 2024-04-28:
Hi 
Sorry, As I have already a pi 5. I was going to use the pi 5 as the GPS linked central time server  creating the PTP time stamps.
 
Then follow your cm4 work for clients which generate remote PPS to drive equipment where GPS is not possible ?

Hope that makes more sense

jclark, 2024-04-28:
So you're trying to use the Pi 5 as the PTP grandmaster (server)? That's not going to work well, because you there will be no hardware timestamping of the PPS signal from the GPS receiver.

closed 2024-04-28 by jclark

g4tsn, 2024-04-28:
Ah ok I understood that the pi5 had the PTP hardware time stamping capable ?

jclark, 2024-04-28:
There are two things that can be hardware timestamped: packets and pulses. Most PTP hardware just does hardware timestamping of packets. What's special about the CM4 (and what you need for a grandmaster) is hardware timestamping of *pulses* (from the GPS receiver). That requires a sync pin (such as provided by the CM4). The post you reference is talking about timestamping *packets*.

g4tsn, 2024-04-28:
Thanks I will go for all CM4 then