jclark/rpi-cm4-ptp-guide #35: testptp pulse width increase beyond 4095 nano seconds ?

g4tsn, 2024-07-11:
Hi 

Is it possible to increase alter the testptp program or something under it to allow a longer pulse from pps out pin on the cm4 ?

 -w 4095 is the maximum which works ....from what I can see the testptp supports larger values ...however this is the maximum value allowed with this hardware without error  ....is there anyway to increase this please ?

maybe there is good reason this cannot be done ?

Thanks 

Jon G4TSN

g4tsn, 2024-12-11:
didnt really get any suggestions here was hoping for a software solution, however a monostable in hardware edge triggered to provide a 100mS pulse seems to work well and produce a PPS similar to most GPS output

jclark, 2025-02-13:
AFAIK it's a hardware limitation

closed 2025-02-13 by jclark