jclark/rpi-cm4-ptp-guide #36: SYNC_OUT does not have any signal output

qqzg, 2024-07-13:
I want to measure the pps accuracy of the output from phc. I follow the steps below.
    sudo ./testptp -d /dev/ptp0 -L 0,2
    sudo ./testptp -d /dev/ptp0 -p 1000000000 -w 1000
But SYNC_OUT does not have any signal output. I used an oscilloscope to measure J2 PIN9.
Do you have any idea, thanks

JN19aban, 2024-07-13:
On what version of Raspbian are you testing? 

The positive of oscilloscope should be on J2 pin and the ground on pin 39 (GPIO) or whatever free ground pin available.

qqzg, 2024-07-14:
> On what version of Raspbian are you testing?
> 
> The positive of oscilloscope should be on J2 pin and the ground on pin 39 (GPIO) or whatever free ground pin available.

Raspbian version:
I use the latest kernel code on the branch rpi-6.6.y, and confirm that the bcm-phy-ptp.c file has the pps feature.

The positive pole of the oscilloscope is connected to PIN9 on J2, and the negative pole is connected to the ground.

jclark, 2024-07-14:
The pulse width is very short (you specified 1 microsecond), so you may need to adjust your oscillator settings for it to detect the pulse.

qqzg, 2024-07-14:
> The pulse width is very short (you specified 1 microsecond), so you may need to adjust your oscillator settings for it to detect the pulse.

jclark, Thanks very much for your reply

I tested in the following way and configured GPO16 as the PPS pin input to receive the SYNC_OUT signal（Connect the GPIO 16 and SYNC OUT pins directly ）, but still couldn't receive the PPS signal from SYNC_OUT，As follows：
```
root@raspberrypi:/home/falcon# ./testptp -d /dev/ptp0 -L 0,2
0,2
set pin function okay

root@raspberrypi:/home/falcon# ./testptp -d /dev/ptp0 -p 1000000000 -w 4000
periodic output request okay

root@raspberrypi:/home/falcon# ppstest /dev/pps0
trying PPS source "/dev/pps0"
found PPS source "/dev/pps0"
ok, found 1 source(s), now start fetching data...
time_pps_fetch() error -1 (Connection timed out)
time_pps_fetch() error -1 (Connection timed out)
time_pps_fetch() error -1 (Connection timed out)
time_pps_fetch() error -1 (Connection timed out)

```
GPS second pulse signals can be received through the PPS input pin, as shown below

```
root@raspberrypi:/home/falcon# ppstest /dev/pps0
trying PPS source "/dev/pps0"
found PPS source "/dev/pps0"
ok, found 1 source(s), now start fetching data...
source 0 - assert 1717342902.788809567, sequence: 10929 - clear  0.000000000, sequence: 0
source 0 - assert 1717342903.788824585, sequence: 10930 - clear  0.000000000, sequence: 0
source 0 - assert 1717342904.788842232, sequence: 10931 - clear  0.000000000, sequence: 0
source 0 - assert 1717342905.788861676, sequence: 10932 - clear  0.000000000, sequence: 0
source 0 - assert 1717342906.788875453, sequence: 10933 - clear  0.000000000, sequence: 0
source 0 - assert 1717342907.788896212, sequence: 10934 - clear  0.000000000, sequence: 0
```

qqzg, 2024-07-14:
> > The pulse width is very short (you specified 1 microsecond), so you may need to adjust your oscillator settings for it to detect the pulse.
> 
> jclark, Thanks very much for your reply
> 
> I tested in the following way and configured GPIO16 as the PPS pin input to receive the SYNC_OUT signal（Connect the GPIO 16 and SYNC OUT pins directly ）, but still couldn't receive the PPS signal from SYNC_OUT，As follows：
> 
> ```
> root@raspberrypi:/home/falcon# ./testptp -d /dev/ptp0 -L 0,2
> 0,2
> set pin function okay
> 
> root@raspberrypi:/home/falcon# ./testptp -d /dev/ptp0 -p 1000000000 -w 4000
> periodic output request okay
> 
> root@raspberrypi:/home/falcon# ppstest /dev/pps0
> trying PPS source "/dev/pps0"
> found PPS source "/dev/pps0"
> ok, found 1 source(s), now start fetching data...
> time_pps_fetch() error -1 (Connection timed out)
> time_pps_fetch() error -1 (Connection timed out)
> time_pps_fetch() error -1 (Connection timed out)
> time_pps_fetch() error -1 (Connection timed out)
> ```
> 
> GPS pps pulse signals can be received through the PPS input pin, as shown below
> 
> ```
> root@raspberrypi:/home/falcon# ppstest /dev/pps0
> trying PPS source "/dev/pps0"
> found PPS source "/dev/pps0"
> ok, found 1 source(s), now start fetching data...
> source 0 - assert 1717342902.788809567, sequence: 10929 - clear  0.000000000, sequence: 0
> source 0 - assert 1717342903.788824585, sequence: 10930 - clear  0.000000000, sequence: 0
> source 0 - assert 1717342904.788842232, sequence: 10931 - clear  0.000000000, sequence: 0
> source 0 - assert 1717342905.788861676, sequence: 10932 - clear  0.000000000, sequence: 0
> source 0 - assert 1717342906.788875453, sequence: 10933 - clear  0.000000000, sequence: 0
> source 0 - assert 1717342907.788896212, sequence: 10934 - clear  0.000000000, sequence: 0
> ```
As shown in the figure below, I found that during the process, GPO16 can occasionally receive a SYNC_OUT PPS signal
[image]

tswaehn, 2025-06-18:
hi, 
I am having issues seeing the SYNC_OUT signal. Any ideas are more than welcome, because I dont have any clue what I am doing wrong here.

my setup:

* CM4 
* [board](https://www.waveshare.com/product/raspberry-pi/boards-kits/compute-module-4-4s-cat/compute-module-4-poe-board.htm) and [schematics](https://files.waveshare.com/upload/a/af/Compute_Module_4_PoE_Board_SchDoc.pdf) 
* the board uses Pin 18 of CM4 (ETHERNET_SYNC_OUT) to Jumper P14.8
* this should probably be the same like the [original rpi](https://datasheets.raspberrypi.com/cm4io/cm4io-datasheet.pdf) page 12 and J2 Pin 9

* using  std raspbian `Linux 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64` 

```
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

```

ptp4l is running:

```
ptp4l[1887.588]: master offset         15 s2 freq   -6335 path delay      1089
```

```
sudo ./testptp -d /dev/ptp0 -L 0,2
sudo ./testptp -d /dev/ptp0 -p 1000000000 -w 1000
```