jclark/rpi-cm4-ptp-guide #38: Compute Module 5 experience

JN19aban, 2024-12-17:
Hi,

I know that is too early to test / have a Compute Module 5 but I see that has exposed the sync out / sync in pin and also support PTP. 
I do not know if the PHY drivers support unicast delays as is on CM4, because on RPI5 only multicast delays supported..... so a test would highly appreciated.

If someone have the CM5 or plan to buy it let as know if support PTP unicast delays.

Also if someone has a solution how to use the Network manager hook instead of dhcpd would be grate.

Regards,
Jim

jclark, 2024-12-17:
I have a CM5.

Most important thing I have found out so far is that the silkscreen on the CM5 IO board is wrong: SYNC_OUT is pin 6 not pin 9.

The PHY on the RPi 5 (BCM54213) is different from the CM5: it doesn't support hardware timestamping. RPi 5 hardware timestamping is supported via the MAC driver.

I wouldn't expect any difference between the CM5 and CM4 as regards PTP support since they have exactly the same PHY (the BCM54210PE), but I haven't tested yet.

JN19aban, 2024-12-17:
Hi @jclark,

That is why I am seeing on RPI5 that have hardware support on PHY and I was wondering what is going on, but as you say is MAC driver.

Here is the output of the RPI5:

```
Time stamping parameters for eth0:

   Capabilities:
        hardware-transmit
        software-transmit
        hardware-receive
        software-receive
        software-system-clock
        hardware-raw-clock
  PTP Hardware Clock: 0
  Hardware Transmit Timestamp Modes:
        off
        on
        onestep-sync
  Hardware Receive Filter Modes:
        none
        all
```

If you have time give a try on CM5 to see if have any difference.

Best regards,
Jim

closed 2025-04-06 by JN19aban