jclark/rpi-cm4-ptp-guide #39: CM5 does not work with ptp4l

jclark, 2025-01-15:
The CM5 doesn't work with ptp4l.  Here's how to test without needing any special hardware.

1. Connect a CM4 and a CM5 directly with a short ethernet cable. When running headless, I do this by having the CM4 use DHCP on eth0, and then on the CM5 plug an ethernet dongle in the USB port and then plug that into my main network; on the CM5 I then use nmcli to configure eth0 with the shared method.
2. Set up chrony on both the CM4 and CM5. Ideally configure with a common local NTP server (by specifying e.g. `server ntp.lan iburst` in `/etc/chrony/chrony.conf`.
3. On the CM5, run phc2sys to synchronize the PHC to the system clock: `sudo phc2sys -q -m -l 6 -c eth0 -s CLOCK_REALTIME -O 0 >phc2sys.log &`
4. Do `tail -f phc2sys.log` and wait for 30 seconds or so until the sys offsets are consistently small (absolute value < 100).
5. Start ptp4l on the CM5: `sudo ptp4l -i eth0 --tx_timestamp_timeout 100 -l 6 -m -q`. After a few seconds it should display `selected local clock 2ccf67.fffe.c114d2 as best master`, where the first and last part of the hex number shown corresponds to the MAC address of eth0 on the CM5 (`2c:cf:67:c1:14:d2` in this case`).
6. Now open a terminal on the CM4 and run ptp4l as a slave: `sudo ptp4l -i eth0 --tx_timestamp_timeout 100 -l 6 -m -q -s >ptp4l.log &`
7. Do `tail -f ptp4.log` and wait till master offset has settled down (absolute value < 100).
8. Now do `phc_ctl eth0 cmp`. This is the difference in nanoseconds between the system clock and the PHC. If everything is working properly this should certainly be less than a millisecond ie 1,000,000. The value I actually see varies all the time. Right now it is `4629122817ns` which is 4.5 seconds.  However, I get wildly different results on different days.

Since the CM4 is known good, this is going to be a hardware, firmware or driver problem on the CM5.

Next step is to investigate with wireshark.

JN19aban, 2025-01-15:
@jclark let me ask something. On CM5 what bootloader it has? I am telling this because on latest updates NUMA is enabled and this might have a drawback on ptp4l.

https://github.com/raspberrypi/rpi-eeprom/tree/master/firmware-2712/latest

Have a check on this.....

jclark, 2025-01-15:
@JN19aban It looks like NUMA can be turned off by setting SDRAM_BANKLOW=0 with rpi-eeprom-config. I will give that a try.

JN19aban, 2025-01-15:
@jclark update to latest first and then disable NUMA

JN19aban, 2025-01-15:
It has to be on this release.... 
```

BOOTLOADER: up to date
   CURRENT: Wed Jan  8 17:52:48 UTC 2025 (1736358768)
    LATEST: Wed Jan  8 17:52:48 UTC 2025 (1736358768)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
```

jclark, 2025-01-15:
`vcgencmd bootloader_version` gives me

```
2025/01/08 17:52:48
version 97facbf492c43a5b6b0e9719860798b7cebfdebb (release)
timestamp 1736358768
update-time 1736908130
capabilities 0x0000007f
```

What command were you running?

jclark, 2025-01-15:
With that the offset is `344096797ns` i.e. 0.34s, which is a lot different from before.

Do you have any insight as to why NUMA should be affecting ptp4l so much?

JN19aban, 2025-01-15:
This command:

`sudo rpi-eeprom-update
`

To update use this:

`sudo rpi-eeprom-update -a`

But you have to change the bootloader release from stable to latest.

To do this you have to run:

```
sudo apt update
sudo apt upgrade
sudo reboot
```

1.` sudo raspi-config`
2. `6 Advance Options`
3. `A5 Bootloader Version`
4. `E1 Latest`
5. Hit ok and the finish and reboot
6. run `sudo rpi-eeprom-update -a`
7. and reboot

jclark, 2025-01-15:
I'm running latest already and get the same output from rpi-eeprom-update as you.

JN19aban, 2025-01-15:
"Do you have any insight as to why NUMA should be affecting ptp4l so much? "

Wrong RAM timings the first one......... and you have to be with NUMA enabled on kernel... otherwise bad very bad performance...

Although I do not know what effect has on PTP4L because yet I do not have available any CM5.

JN19aban, 2025-01-15:
So this value `344096797ns` is with NUMA enabled or disabled you posted before?

jclark, 2025-01-15:
With NUMA disabled

JN19aban, 2025-01-15:
OK try with `SDRAM_BANKLOW=1`

and re run the test with `SDRAM_BANKLOW=2`

jclark, 2025-01-15:
The first result (more than 4 seconds offset) was with latest firmware and nothing added to the firmware config, so NUMA enabled.

JN19aban, 2025-01-15:
Theoretically is with `SDRAM_BANKLOW=3` but to make sure run with 1 and 2 values......

JN19aban, 2025-01-15:
To add something important. After all the test try with `SDRAM_BANKLOW=-1` to disable NUMA and RAM enchantments.

JN19aban, 2025-01-15:
@jclark I am thinking something more.

With the NUMA setup (RAM timings etc.) the system is adjusted for performance, compatibility, etc.

On theory the HW Timestamps shouldn't be affected that much as the previously posted latencies.

Can you do more testing that is the HW Timestamp truly enabled and not "faked" somehow on PTP4L? and the whole latency you see might be Software Timestamps?

An other scenario I am thinking can be the new I/O controller (RP1) that might affect somehow the performance of the PHY???

jclark, 2025-01-15:
With SDRAM_BANKLOW=1, I get -26605883ns (so -26,605,883ns = 0.02s).

JN19aban, 2025-01-15:
It might be, I will catch an eye for more news that come up on the RPI 5 / CM 5. I will probably have one on my hands (CM5) by the 15th of February if all goes well.

I strongly think these 2 things on my previous comment though.

jclark, 2025-01-16:
Wireshark shows clearly that the problem is that the hardware transmit timestamps from the CM5 are incorrect.

After some quality time with  bpftrace, I think I know what is going on. The behaviour is a kernel bug. It is a result of three things:

1.  the CM5 has two PTP clocks associated with eth0, each with its own packet timestamper; one is the PHY level one from drivers/net/phy/bcm-phy-ptp.c and the other is MAC level one from drivers/net/ethernet/cadence/macb_ptp.c (neither the CM4 nor Pi 5 have two clocks)
2.  the macb driver uses a [legacy method](https://github.com/raspberrypi/linux/blob/dfff38316c1284c30c68d02cc424bad0562cf253/drivers/net/ethernet/cadence/macb_main.c#L4000) of controlling hardware timestamping, using ndo_eth_ioctl instead of the newer ndo_hwtstamp_set/_get methods
3. the code that handles these ioctls (dev_{set,get}_hwtstamp) in net/core/dev_ioctl.c doesn't properly handle the case whether there are both MAC and PHY level timestampers, but the MAC level timestamper uses the legacy ndo_eth_ioctl

The overall result is that the hardware transmit timestamping code in the PHY level driver doesn't get called, and I think there is a weird mix of the two PHCs being used, which is why there are wrong timestamps.

In kernel 6.8, the macb driver is updated to use the newer ndo_hwtstamp_set/_get, so this problem should go away.  I tried applying the [patch for this](https://github.com/torvalds/linux/commit/202cb220026e4dabb592852e7555e264656445ac) to the current Raspberry Pi kernel, but I got strange errors from the macb driver ("DMA bus error: HRESP not OK"), and I have absolutely no idea what is causing those.

The `next` branch of the raspberry pi kernel (which can be installed with `rpi-update next`) is based on 6.12, so has the updated driver. Alternatively compiling the current kernel commenting about `CONFIG_MACB_USE_HWSTAMP` in the config also avoids having the two competing clocks.

Unfortunately after fixing this problem, another appears: ptp4l gives the infamous `timed out while polling for tx timestamp` error, even after increasing increasing tx_timestamp_timeout to a ridiculously large value. Argh!

JN19aban, 2025-01-16:
This means that you see on `/dev ` two clocks `ptp0` and `ptp1`?

jclark, 2025-01-16:
Right: the CM5 has a /dev/ptp0 (like the /dev/ptp0 on the CM4) and a /dev/ptp1 (like the /dev/ptp0 on the Pi5).

JN19aban, 2025-01-16:
That is new..... so you can lets say use 2 clocks (an example PTP and SyncE together) of course when will work correct. That sounds nice so far.

lhoward, 2025-01-17:
It sounds nice but isn't the issue described [here](https://lpc.events/event/11/contributions/949/attachments/823/1555/paper.pdf) where the kernel can only report one timestamp at a time?

jclark, 2025-01-18:
I just did an update to `next` (using `sudo rpi-update next`), and it is now working. The test above gives a difference of 0.5 milliseconds, which is about right.

uname -a gives

```
Linux valdeon 6.12.10-v8-16k+ #1840 SMP PREEMPT Fri Jan 17 18:08:09 GMT 2025 aarch64 GNU/Linux
```

rpi-eeprom-update gives

```
BOOTLOADER: up to date
   CURRENT: Tue 14 Jan 00:16:48 UTC 2025 (1736813808)
    LATEST: Tue 14 Jan 00:16:48 UTC 2025 (1736813808)
   RELEASE: latest (/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
```

JN19aban, 2025-01-18:
@jclark So it seems to be fixed on kernel` 6.12.x` Kernel. Nice.

@lhoward See this https://lwn.net/Articles/859792/

geerlingguy, 2025-01-21:
Have you raised an issue about this in the Pi kernel repo? They might be willing to back port the fix. Or maybe motivation to get to next LTS kernel sooner, which I would love for many other reasons!

jclark, 2025-01-21:
@geerlingguy I posted it on the forum thread where they invited feedback on the 6.12 kernel https://forums.raspberrypi.com/viewtopic.php?t=379745&start=100#p2287594. Hopefully that will put it on their radar and encourage them to move to 6.12 soon.

I don't think the backport would be easy. There's the problem with the MAC timestamper being used. The only fix I see for that is to update the macb driver to the new method for setting hardware timestamps. But when I did that, I got mysterious DMA-related errors. Then there's another problem related to the interrupt register in the bcm_phy_ptp driver, and I have no idea what fixed that.

As of Nov 20th, they were talking about moving to 6.12 in a few months, so I'm hoping it will be soon. But `sudo rpi-update next` is already pretty easy.

Waynechen026, 2025-02-12:
Hi @jclark  ,I have updated to 6.12, but I still cannot use ptp4l in CM5. The error "timed out while polling for tx timestamp" still occurs. Why is this happening? We are using the TimeProvider TP4100 for time synchronization.

jclark, 2025-02-12:
@Waynechen026 Are you using the tx_timestamp_timeout option? This is still needed. I usually use 100. If you still get the error, open a new issue with full details of your setup (kernel version, firmware version, logs, hardware etc).

JN19aban, 2025-04-06:
@jclark let me ask something. Have you update to the latest stable kernel 6.12.20? If so are you experiencing any issues so far?

I am saying this because on one of my CM4 that I have updated I am seeing some weird latency spikes which, on my second one still with 6.6.74 kernel the spikes are not there. 

The same spikes are and on RPI 5 with 6.12.20 kernel.

It seems that the PPS performance is terrible on 6.12.20......

geerlingguy, 2025-04-08:
@JN19aban hmm... I thought it might've been my network switch causing the spikes I was seeing on my Pi 5 with an i226... but maybe it's 6.12, I may have to switch back to the older kernel and re-test.

jclark, 2025-04-08:
I'm running 6.12.20 on a CM5 (with the TimeBeat MAX-F10S board). I am not seeing any problems (in terms of offsets between PHC and timestamp pulses). When you say latency spikes, what exactly are you seeing?

BTW, @JN19aban @geerlingguy I made my eventual replacement for this guide public (though without any announcements yet). See https://satpulse.net I think it should work out of the box on the Pi5/i226 hat, but I haven't tried it yet.

JN19aban, 2025-04-08:
Hey guys,

The spikes are random and specially they start after couple of hours. Firstly I noticed that by "accident" when I tried to log in via SSH to the CM4 running the 6.12.20 and then started monitoring the logs. After some time of waiting it suddenly came up the lag spike and almost completely halted the CM4....... To not say more I use a simple script to monitor the system latency depending the CPU governor. 

On the 6.12.20 when the systems boot up everything is normal but slowly it starts the latency to increase until it hit very hard (like page_fault) and then become normal again until happens again and goes on.......

This happens more quicker when the ts2phc is on action.... and takes more time for the lag spike when the ts2phc is disabled.....

Also I am noticing the same thing on the RPI5, which the usage is to receive an audio stream and have a sound card for playback.........

**The most weird of all is this:**

I also have 2 RPI4 rev 1.2 and rev 1.4. On rev 1.2 the lag spikes are there, but on rev 1.4 everything is normal no lag spikes no anything....

To mention this also:

The CM4 that has the 6.12.20 is rev. 1.1 and the second one is rev 1.0 with 6.6.74 kernel which not have the lag issue (I will try later to update to 6.12.y kernel for test).....

For the results of the test I will update in next days cause I am out of home......

@geerlingguy this happens even without the usage of I226??

@jclark this seems to be a cool project, I will try as soon I come back...... one question though, the 1gb version CM4 will be enough for the satpulse to be working ok?

The script I use for latency test is this:

````
#!/bin/bash

# Function to measure timer latency (overhead of getting the time)
measure_timer_latency() {
    local samples=1000
    local min_latency=999999999
    local max_latency=0
    local total_latency=0

    echo "Measuring Timer Latency..."

    for ((i=0; i<samples; i++)); do
        start=$(date +%s%N)
        end=$(date +%s%N)
        latency=$((end - start))

        # Track min, max, and sum for average calculation
        ((latency < min_latency)) && min_latency=$latency
        ((latency > max_latency)) && max_latency=$latency
        total_latency=$((total_latency + latency))
    done

    avg_latency=$((total_latency / samples))
    echo "Timer Latency (nanoseconds):"
    echo "  Min: $min_latency ns"
    echo "  Max: $max_latency ns"
    echo "  Avg: $avg_latency ns"
}

# Function to measure subprocess execution latency
measure_subprocess_latency() {
    local samples=100
    local min_latency=999999999
    local max_latency=0
    local total_latency=0

    echo "Measuring Subprocess Execution Latency..."

    for ((i=0; i<samples; i++)); do
        start=$(date +%s%N)
        echo "test" > /dev/null
        end=$(date +%s%N)
        latency=$((end - start))

        ((latency < min_latency)) && min_latency=$latency
        ((latency > max_latency)) && max_latency=$latency
        total_latency=$((total_latency + latency))
    done

    avg_latency=$((total_latency / samples))
    echo "Subprocess Latency (nanoseconds):"
    echo "  Min: $min_latency ns"
    echo "  Max: $max_latency ns"
    echo "  Avg: $avg_latency ns"
}

# Function to measure GPIO latency (Requires GPIO pin access)
measure_gpio_latency() {
    local pin=17
    local samples=100
    local min_latency=999999999
    local max_latency=0
    local total_latency=0

    echo "Measuring GPIO Latency on Pin $pin..."

    # Export GPIO pin for use
    echo $pin > /sys/class/gpio/export
    echo "out" > /sys/class/gpio/gpio$pin/direction

    for ((i=0; i<samples; i++)); do
        start=$(date +%s%N)
        echo 1 > /sys/class/gpio/gpio$pin/value
        echo 0 > /sys/class/gpio/gpio$pin/value
        end=$(date +%s%N)
        latency=$((end - start))

        ((latency < min_latency)) && min_latency=$latency
        ((latency > max_latency)) && max_latency=$latency
        total_latency=$((total_latency + latency))
    done

    avg_latency=$((total_latency / samples))
    echo "GPIO Latency (nanoseconds):"
    echo "  Min: $min_latency ns"
    echo "  Max: $max_latency ns"
    echo "  Avg: $avg_latency ns"

    # Cleanup GPIO
    echo $pin > /sys/class/gpio/unexport
}

# Run all latency tests
measure_timer_latency
measure_subprocess_latency

# Uncomment this line if you want to test GPIO latency
# measure_gpio_latency
```
`

Regards.

jclark, 2025-04-08:
@JN19aban I haven't tried it on 1Gb model yet. But it's running on a 2gb model, and /proc/meminfo shows a 1gb free, so I think it should be fine provided you're doing too much else. I do have a 1Gb CM4 somewhere, so I should try it to make sure.

JN19aban, 2025-04-09:
OK, I would try out and on my side and hope to be sufficient enough.....

regymm, 2025-05-03:
Hi, thanks for this discussion I successfully set up PTP time syncing between CM5 and a TAP-OCP Time Card, as documented here: https://github.com/regymm/Time-Card-Flex/blob/main/DOC/Setup-and-Usage.md
I'm using `Linux raspberrypicm5 6.12.15-v8-16k+ #1856 SMP PREEMPT Wed Feb 19 14:26:04 GMT 2025 aarch64 GNU/Linux` by updating to `next`. All other setups are default (no NUMA changes). 
I noticed the accuracy of the CM5's time according to PPS output is ~170 ns off from the grandmaster clock. With the same GMC, Intel I210 on RPi5 can sync up to ~20 ns. 
I'm wondering if anyone has related results? Don't know if this is a chip performance limit on CM5 or if it can be optimized somehow.

jclark, 2025-05-03:
@reggym You don't need to update to next any more. Just an apt update is enough.

When you say the accuracy is 170ns, do you mean the difference fluctuates but with an absolute value <170ns, or do you mean it is consistently 170ns different? What do the ptp4l offsets settle down to?

I haven't found the accuracy of the CM4 to be any worse than the i210, and since the CM5 uses the exact same PHY, I would not expect it to be any different. So I am not sure what the issue is.

I am interested that you have the i210 working on the RPi5. What OS are you using? Did you have to recompile the kernel?

geerlingguy, 2025-05-03:
I have also not yet started testing with a CM5, but hope to later this year. I'm still running all my current tests on a Pi 5, but using an external PCIe NIC (Intel i226) since the Pi 5 doesn't have PPS exposed for the NIC (wish we had at least a test point!).

regymm, 2025-05-04:
Oh I mean the fluctuation is small (~20 ns or lower), but the CM5 PPS has a constant ~170 ns offset with GMC PPS. Ptp4l settles down to ~zero offset. 

With the exact same setup, an i210 on RPi5 shows ~20 ns offset when stabilized. I'm using stock Raspbian kernel on RPi 5, and it just works out of the box after PCIe is enabled. No kernel modification.

jclark, 2025-05-04:
If the latency is constant, then you may be able to reduce it  by tweaking ptp4l ingressLatency and/or egressLatency options.

I think there is some delay in the perout process, but only a few nanoseconds.