jclark/rpi-cm4-ptp-guide #40: The CM5 have the error "timed out while polling for tx timestamp" when using ptp4l.

Waynechen026, 2025-02-12:
I cannot use ptp4l on CM5 4GB. No matter how I adjust the tx_timestamp_timeout option, it still shows "timed out while polling for tx timestamp."

I have already upgraded the kernel to 6.12.x using `sudo rpi-update next`.  

`uname -a` gives:
`Linux raspberrypi 6.12.10-v8-16k #1 SMP PREEMPT Thu Jan 23 00:35:01 CST 2025 aarch64 GNU/Linux`

then using `sudo rpi-eeprom-update -a` , it gives
```
BOOTLOADER: up to date
   CURRENT: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
    LATEST: Wed 22 Jan 00:16:51 UTC 2025 (1737505011)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
```
In our laboratory setup, we use the TimeProvider 4100 for time synchronization and the Netgear S308 switch to connect the embedded system.

jclark, 2025-02-12:
What ptp4l command are you using? And what is the config file (if any)? Please also attach the relevant output from ptp4l.

jclark, 2025-02-12:
I just did another `rpi-update next`, and this puts me on

```
Linux raspberrypi 6.12.13-v8-16k+ #1848 SMP PREEMPT Mon Feb 10 13:18:35 GMT 2025 aarch64 GNU/Linux

BOOTLOADER: up to date
   CURRENT: Wed 12 Feb 10:51:52 UTC 2025 (1739357512)
    LATEST: Wed 12 Feb 10:51:52 UTC 2025 (1739357512)
   RELEASE: latest (/usr/lib/firmware/raspberrypi/bootloader-2712/latest)
            Use raspi-config to change the release.
```

Can you also update again so we are running the same versions ? Then I can attempt to reproduce (once I known your ptp4l setup).

Waynechen026, 2025-02-12:
@jclark I am using linuxptp version 4.4 with the default configuration file. 
[default.txt](https://github.com/user-attachments/files/18768780/default.txt)

The command I run is: `sudo ptp4l -m -H -i eth0 --tx_timestamp_timeout 100`

In kernel 6.6.51, we also have the problem where ptp4l show the error message:

"received DELAY_REQ without timestamp."

We fixed this issue by running the following command: `sudo ptp4l -m -H -i eth0 --hwts_filter "full"`

However, the synchronized time seems to be incorrect, and the time drift keeps increasing.

[image]

In CM4, we use LinuxPTP 4.0, and synchronization works correctly.

jclark, 2025-02-12:
You will need ptp_minor_version 0 option with LinuxPTP 4.x.

Waynechen026, 2025-02-12:
Thank you !! I found my problem.

closed 2025-02-12 by jclark

Waynechen026, 2025-02-13:
I downloaded the files from [[https://github.com/richardcochran/linuxptp](https://chatgpt.com/c/url)](url) and ran `make & make install`.
 I initially thought that modifying "default.cfg" and running "ptp4l" would work directly, but that was not the case, I had to explicitly specify the configuration file.