jclark/rpi-cm4-ptp-guide #41: CM5 never settling

Deadevilgrounds, 2025-04-12:
So I have a CM5 module with the standard CM5 IO board and case.
I don't have any fancy GPS modules yet, but I have an Adafruit module based on the MTK3339, and I also have a knockoff u-blok 7 module.
I have tried both and I get the same result.

What I am seeing is that the master offset never seems to settle down. I have seen it go as low as 4ns or 16ns from a run of about 20 seconds, then it will explode and go crazy and oscillate between values, dropping out of s2 back to s0, etc.

I also see constant ptp4l messages about master sync timeout and master tx announce timeout.

Here is a Tmux window showing the three main components

[image]

[image]

[image]

I was initially worried that the PPS was not working well so I also enabled one of the GPIO pins and plugged it into that and I can see it pulsing once per second regularly. Some of those jumps in the master offset are over a second long so I would expect to see that even watching the GPIO trigger.

I do have a nicer GPS unit at work, both an Evertz and a Meinburg, as we use them for ST2110 video delivery, but this is just my home project. 

I have run through the md files in this repo, there was a typo in the phc2shm.service. 
Line 4 refers to the ts2phc service but its missing ".service". So I modified that line to 
`Requires=ts2phc.service
`
I also made a small change to the ptp4l-gm-set.sh script so that it won't try to set the pmc params until it detects ptp4l; otherwise, those params won't stick. So I added this before the pmc call
```
# Wait for ptp4l UDS socket to appear
for i in {1..10}; do
  if [ -S /var/run/ptp4l ]; then
    break
  fi
  echo "Waiting for ptp4l to be ready..."
  sleep 0.5
done
```

Ive been reading through the other open tickets and I've tried setting the SDRAM_BANKLOW=1 from #39 
Ive also updated my firmware to the latest and I'm on the latest kernel.

[image]
[image]

I know the GPS module is cheap but not 1s variance cheap lol. 

So if anyone has any ideas for me to try that would be great.

Deadevilgrounds, 2025-04-12:
[image]

There is an example of when its all happy and with low offsets. Its seems very random when it becomes good or not.

Deadevilgrounds, 2025-04-12:
[image]

Then a few seconds/min later its back to out of bounds.

Deadevilgrounds, 2025-04-14:
Ok so it came down to message types/rates.
For the MTK3339 the default set of messages from cold boot + factory default is
$PMTK314,1,1,1,1,1,5,0,0,0,0,0,0,0,0,0,0,0,0,0*2C<CR><LF> 
which equates to 
0 NMEA_SEN_GLL, // GPGLL interval - Geographic Position - Latitude longitude 
1 NMEA_SEN_RMC, // GPRMC interval - Recommended Minimum Specific GNSS Sentence
2 NMEA_SEN_VTG, // GPVTG interval - Course over Ground and Ground Speed 
3 NMEA_SEN_GGA, // GPGGA interval - GPS Fix Data 
4 NMEA_SEN_GSA, // GPGSA interval - GNSS DOPS and Active Satellites 
5 NMEA_SEN_GSV, // GPGSV interval - GNSS Satellites in View

closed 2025-04-14 by Deadevilgrounds

jclark, 2025-04-14:
So the problem was the GPS transmitting more data than the baud rate could accomodate?

Deadevilgrounds, 2025-04-14:
I believe so.
I had a little look at the linuxptp code to see which messages were used for time and just picked the RMC one.
I haven't done any calculations on what the 9600 baud could send vs what it was trying to send. 

9600 baud would be about 960 ASCII chars per second so maybe???

It would tie in to why the problem would come and go over time and why using an external antenna made it worse, as the more fixes I have, the more extended the message queue would be per second. Ironically having it in my window made it better lol.

Since I dont need this module for location data, only time this works well. Im now getting constant -5 to 5 ns offsets and its stable.

So I can confirm it works well on a CM5 with the Adafruit V3 using the MTK3339. I also just nicked...borrowed a MTK3333 from someone I work with and it also works great.